REhints/efiXplorer #1: Subprocess Error

TylerForester, 2020-09-10:
C:\python37-x64>python.exe C:\Users\Pavel\Desktop\efiXplorer-master\efiXplorer-master\build.py "C:\Program Files\IDA Pro 7.5"
Traceback (most recent call last):
  File "C:\Users\Pavel\Desktop\efiXplorer-master\efiXplorer-master\build.py", line 55, in <module>
    build()
  File "C:\python37-x64\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\python37-x64\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\python37-x64\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\python37-x64\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\Pavel\Desktop\efiXplorer-master\efiXplorer-master\build.py", line 34, in build
    '-DIdaSdk_ROOT_DIR={}'.format(idasdk_dir),
  File "C:\python37-x64\lib\subprocess.py", line 339, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\python37-x64\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\python37-x64\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified

yeggor, 2020-09-10:
Hi, Pavel, make sure, that build tools installed (CMake + Visual Studio 2019). It looks like the script cannot execute `cmake`.
Also, it should be noted that the argument is not the path to the IDA, but the path to the idasdk.
As a last resort, you can download the latest builds here: https://github.com/binarly-io/efiXplorer/actions/runs/242276280. But don’t forget about the `guds` directory. It should also be placed in the `<IDA_DIR>/plugins` directory.

yeggor, 2020-09-10:
Anyway, thanks for the issue, we should add exception handling.

TylerForester, 2020-09-11:
Thank! The finished build helped. But during the tests, it was noticed that not all services are marked up.

yeggor, 2020-09-11:
can you share the efi images you tested?

TylerForester, 2020-09-11:
> can you share the efi images you tested?

Which I can't test.  If I find something similar, I will provide it.

TylerForester, 2020-09-12:
Found a similar issue in another efi file. FreePool should be located at the address below, but it is not marked up.

Address: 0xB70

File: [https://yadi.sk/d/iSsHsc7tkq-L5A](url)

yeggor, 2020-09-12:
Yes i can see it, thanks. I know what the problem is. Will fix it soon.
But since gBS is defined correctly, the pseudocode is fine:
```c
{
  ...
  else
  {
    v4 = 0x8000000000000009ui64;
  }
  (gBS_1E468->FreePool)(v11);
  return v4;
}
```

TylerForester, 2020-09-14:
We are waiting for the update.

> Yes i can see it, thanks. I know what the problem is. Will fix it soon.
> But since gBS is defined correctly, the pseudocode is fine:
> 
> ```c
> {
>   ...
>   else
>   {
>     v4 = 0x8000000000000009ui64;
>   }
>   (gBS_1E468->FreePool)(v11);
>   return v4;
> }
> ```

We are waiting for the update.

cifamcg, 2020-12-24:
C:\Program Files\IDA Pro 7.5\loaders\efiXplorer64.dll: not IDA DLL file

yeggor, 2020-12-24:
Hi, to install the plugin and loader correctly you need to do the following:
* copy the file `efiХloader64.dll` in the directory `C:\Program Files\IDA Pro 7.5\loaders\` (not `efiXplorer64.dll`)
* copy the `guids` directory and files `efiXplorer.dl`, `efiXplorer64.dll` to the directory `C:\Program Files\IDA Pro 7.5\plugins\`

If you install everything correctly, when you open the firmware image in IDA, you will see the following:
[image]
For analysis go to `edit->plugins->efiXplorer` (`Ctrl+Alt+E`)

Besides, you can open UEFI drivers (not the whole firmware image) and analyze them, you don't need efiXloader for this.

cifamcg, 2020-12-24:
[image]
[image]
[image]
[image]

cifamcg, 2020-12-24:
[image]

cifamcg, 2020-12-24:
[image]

yeggor, 2020-12-24:
Copy the file **efiХloader64.dll** in the directory `C:\Program Files\IDA Pro 7.5\loaders\` (not **efiXplorer64.dll**).
The difference is that **efiXplorer** is an IDA plugin and **efiXloader** is a UEFI firmware loader.
You can get efiXloader64.dll binary from release attach: `\efiXloader\efiXloader64.dll`.
Or you can build it from sources: 

```bash
cd efiXloader
mkdir build
cd build
cmake .. -DIdaSdk_ROOT_DIR="/path/to/idasdk"
cmake --build . --config Release
```

cifamcg, 2020-12-25:
think you work perfect

closed 2020-12-25 by matrosov