REhints/efiXplorer #3: Crash in EFILoader

alexander-pick, 2020-10-14:
I am experiencing a crash trying to load an EFI firmware into IDA with the latest version. I compiled the module for Linux and Windows and tried the precompiled version as well all with the same result. I also tried IDA 7.4 and 7.5 for comparison without success. The module crashes IDA during analysis, IDA simply closes itself and creates a dmp. 

Now I am a bit unsure what the best way to report this, I can provide the EFI image which I am trying to load if this helps.

alexander-pick, 2020-10-15:
[R70.zip](https://github.com/binarly-io/efiXplorer/files/5386748/R70.zip)

It is a complete EFI firmware image with lot of modules. The loader parses it to a certain point and than IDA closes itself. The file is attached above.

matrosov, 2020-10-16:
Thank you for the bug submission. We will investigate this problem.

yeggor, 2021-05-04:
Tested on the latest version, everything works (efiXloader + efiXplorer).
[image]

closed 2021-05-04 by yeggor