REhints/efiXplorer #5: efi rom "doesn't contain any instructions or data" when using efiXloader

hanetzer, 2021-04-12:
Pretty much says it all. This is the latest bios update for my pc, tried it both with and without the uefi cap header.
Using the BHEU release on this repo.
File is available here: https://rog.asus.com/us/motherboards/rog-crosshair/rog-crosshair-vii-hero-wi-fi-model/helpdesk_bios
[image]

hanetzer, 2021-04-12:
actually a bit more. Manually running the plugin (ctrl+alt+e) gets this output, but no changes thereafter.
```
[efiXplorer] ========================================================
[efiXplorer] plugin run
[efiXplorer] guids.json exists: true
[efiXplorer] input file is UEFI firmware
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] ImageHandle finding
[efiXplorer] ========================================================
[efiXplorer] SystemTable finding
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] SMST finding
[efiXplorer] ========================================================
[efiXplorer] Boot services (protocols):
+-----------+-----------+
|  Address  |  Service  |
+-----------+-----------+
[efiXplorer] ========================================================
[efiXplorer] finding of other addresses of global gBS variables
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] SmmServices finding (all)
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] ========================================================
[efiXplorer] protocols printing
[efiXplorer] ========================================================
[efiXplorer] protocols marking
[efiXplorer] SwSmiHandler function finding (using EFI_SMM_SW_DISPATCH(2)_PROTOCOL_GUID)
[efiXplorer] ========================================================
[efiXplorer] Looking for SMM callout
[efiXplorer] ========================================================
[efiXplorer] Looking for GetVariable stack/heap overflow
[efiXplorer] less than 2 GetVariable calls found
[efiXplorer] ========================================================
[efiXplorer] Looking for SmmGetVariable stack/heap overflow
[efiXplorer] gSmmVar->SmmGetVariable calls finding via EFI_SMM_VARIABLE_PROTOCOL_GUID
[efiXplorer] can't find gSmmVar addresses
[efiXplorer] less than 2 GetVariable calls found
[efiXplorer] ========================================================
[efiXplorer] Looking for EFI_SMM_CPU_PROTOCOL
[efiXplorer] can't find gSmmCpu addresses
[efiXplorer] ========================================================
[efiXplorer] the log is saved in a JSON file
```

yeggor, 2021-04-12:
The problem is that there are no UI sections with EFI image names in this firmware. Therefore, our bootloader does not extract EFI drivers.
Alternatively, use the uefi_retool utility to extract the EFI images, after which you can parse each one individually with IDA + efiXplorer.

To get images with uefi_retool:

```
git clone https://github.com/yeggor/uefi_retool.git
cd uefi_retool
python -m pip install requirements.txt
python uefi_retool.py get-images {FIRMWARE_PATH}
```

hanetzer, 2021-04-13:
Hrm. So the issue is with this particular rom's setup, then? If I use your second recommended option, will
I still be able to get all the benefits of an efiXloader'd image?

yeggor, 2021-04-13:
you can run the eXplorer plugin on each separate image, but the loader will not work in this case

hanetzer, 2021-04-13:
I mean, getting access to all images within a single ida session

yeggor, 2021-04-13:
Correct fix takes time, at the moment I added the ability to load images from a directory (https://github.com/binarly-io/efiXplorer/commit/8ba601fde00d1e4984d727f5ccb88630e852f43f).

If your firmware is named "ROG-CROSSHAIR-VII-HERO-WIFI-ASUS-4301.CAP", create a directory next to it "ROG-CROSSHAIR-VII-HERO-WIFI-ASUS-4301.CAP.i64.efiloader" and copy to this directory all the images you want to load.

Then everything is the same: open with a loader, wait for the automatic analysis to complete and run the plugin.

yeggor, 2021-04-13:
Version of the efiXloader that I tested on your firmware: [efiXloader.zip](https://github.com/binarly-io/efiXplorer/files/6300946/efiXloader.zip)

hanetzer, 2021-04-13:
yep, your binary works for me, when doing the above steps. Quite cool.

closed 2021-08-02 by yeggor