REhints/efiXplorer #7: lack support for arm/aarch64

hiberhe, 2021-05-13:
efixplorer only supportsefi files/roms compiled for ia32/x64, it doesn't support efi files/roms compiled arm/aarch64.

matrosov, 2021-05-14:
yep, that's correct. We plan to add support for ARM. Any chance you can share some firmware images for testing purposes?

hiberhe, 2021-05-27:
See the Attached files for UEFI Firmware Image and EFI files compiled for AARCH64(ArmVirtQemu). They ware built using edk2-stable202102 and compressed by 7z(https://www.7-zip.org/).

[ArmVirtQemu_FV.zip](https://github.com/binarly-io/efiXplorer/files/6553595/ArmVirtQemu_FV.zip)
[ArmVirtQemu_EFI_Files.zip](https://github.com/binarly-io/efiXplorer/files/6553600/ArmVirtQemu_EFI_Files.zip)

NewDwarf, 2021-10-30:
That would be nice to add aarch64 support to be able to reverse engineer the android bootloaders.

NikolajSchlej, 2022-07-13:
An aarch64 EFI image can be extracted from [Lenovo update](https://support.lenovo.com/us/en/downloads/ds556845-bios-update-utility-bootable-cd-for-windows-11-thinkpad-x13s-gen-1-type-21bx-21by) using innoextract.

NikolajSchlej, 2022-10-07:
Should this be closed as complete, @yeggor?

yeggor, 2022-10-07:
Yes, thank you. I'm closing that issue for now. AArch64 modules analysis needs to be improved over time. But I suppose we will use other issues for future requests.

closed 2022-10-07 by yeggor