REhints/efiXplorer #8: check unsupporting processors, please!

Raven19409, 2021-05-17:
I load IDA base for PowerPc and see: 

**_[many different dashes & lines]_**
`        __ ___   __      _`
`       / _(_) \ / /     | |`
`   ___| |_ _ \ V / _ __ | | ___  _ __ ___ _ __`
`  / _ \  _| | > < | '_ \| |/ _ \| '__/ _ \ '__|`
` |  __/ | | |/ . \| |_) | | (_) | | |  __/ |`
 ` \___|_| |_/_/ \_\ .__/|_|\___/|_|  \___|_|`
 `                 | |`
`                  |_|`
**(c) 2020, Binarly - https://github.com/binarly-io/efiXplorer**

**Question:**
You can processing UEFI modules for PowerPC?
May be correctly check processors you plugin can support and not loading it for others!

matrosov, 2021-05-17:
We currently support x86 only but planning to add IR-based heuristics to support more architectures naturally. Stay tuned!

I will appreciate if you can share power-pc samples for testing.

Raven19409, 2021-05-18:
Да речь в данном случае совсем не об UEFI и не возможностях вашего плагина!
Стандартно для плагинов Иды при загрузке идёт проверка, на тип процессора,
код для которого может данный плагин обрабатывать, в противном случае
плагин не загружается!
Я работаю с обычными PowerPC файлами, так что об UEFI в данном случае и
речи нет.
Абсолютно непонятно, зачем ваш плагин загружается в этом случае.
Инфа о его загрузке в Output window лично мне буквально глаз "рэ-жэт", ну
не должно ей там быть, плагин не должен болтаться в памяти в этом случае.
И это минимальный уровень защиты от всяких "неожиданностей", кот. возможны
при запуске и работе Иды!!!
(Кстати, хотя в предыдущей "жизни" активно занимался
дизассемблированием/декомпиляцией UEFI c 2015 по март 2019, ни разу не
слышал об UEFI на PowerPC, тем более не встречал "живых" экземпляров. Вроде
бы и в Tianocore Edk2 таких примеров не было, по крайней мере год назад!

closed 2021-05-18 by matrosov