REhints/efiXplorer #11: Overwriting drivers on extraction

ystrelkov, 2021-08-03:
For some firmware, say HP ZBook Studio, the EfiLoader overwrites drivers on extraction.
uefitool.cpp line 59 utf16_utf8(&module_name, reinterpret_cast<const wchar16_t *>(file-uname.data())). But virtually ALL drivers in the firmware have duplicated text names (data) (say 0127  is used by two different drivers). As the result the IDA is missing nearly a HALF of the drivers!!! Took me some time to realize that. Hard to detect and very easy to fix (I've created a getUniqueName function which returns 0127_1, 0127_2, etc..)

yeggor, 2021-08-03:
Yes, this is a significant point, thank you

yeggor, 2021-08-03:
@ystrelkov, can you send an example of the firmware you tested on?

ystrelkov, 2021-08-04:
Here is the 64 bit firmware from (16mb chip) ZBook Studio G5. Nearly all the drivers have duplicate "Text" (in UEFITool). Just search for, say 0127 and see it yourself (one is DXE driver another is SMM module). BTW it's a very nice tool you've created!!!!
[16 orig np.zip](https://github.com/binarly-io/efiXplorer/files/6928128/16.orig.np.zip)

ystrelkov, 2021-08-04:
I guess my fix was a very simple one (I was using a set). Here we go (in case you're interested).
Best regards,
Yuri.
[uefitool_fix.zip](https://github.com/binarly-io/efiXplorer/files/6928208/uefitool_fix.zip)

yeggor, 2021-08-04:
Thank you very much, already fixed (the same thing, only with vector: https://github.com/binarly-io/efiXplorer/commit/15ce9b3bef28711466fc567731104f22ed29654a), but I agree that a set is more suitable here.

closed 2021-08-04 by yeggor