REhints/efiXplorer #38: Problems with duplicated GUIDs

assafcarlsbad, 2022-03-25:
Hi,

I recently encountered an SMM image that for some reason had two copies of `EFI_SMM_BASE2_PROTOCOL_GUID`. As can be seen in the image below, `efiXplorer` renamed the first object to `EFI_SMM_BASE2_PROTOCOL_GUID` and the second to `EFI_SMM_BASE2_PROTOCOL_GUID`_**0**. 
[image]

Later, `EFI_SMM_BASE2_PROTOCOL_GUID`_**0** is passed to `LocateProtocol` but `efiXplorer` doesn't set the type of the corresponding interface (`qword_1E498`). Is it caused by the "non-standard" protocol name?
[image]

Attached is the problematic binary.
[NvramSmm.zip](https://github.com/binarly-io/efiXplorer/files/8346439/NvramSmm.zip)

Thanks!

yeggor, 2022-03-25:
Hello. I've just checked version from master, it seems to be working for me.
You may need to refresh the pseudocode with F5 (or refresh the pseudocode and perform rescan).
[image]

yeggor, 2022-03-25:
@assafcarlsbad
btw, efiXplorer now names child SW SMI handlers `ChildSwSmiHandler` instead of `SmiHandler`.
So you should probably fix it here: https://github.com/Sentinel-One/brick/blob/12e4172c2a1ae92252f8cca763374926a3b1411f/brick/ida/modules/efiXplorer/efiXplorer.py#L23

assafcarlsbad, 2022-03-25:
Just verified it, the `master` build indeed fixes this issue.
And thanks for the heads up regarding the name change!

closed 2022-03-25 by assafcarlsbad