REhints/efiXplorer #56: IDA crashes after analyzing firmware sample with efiXplorer latest builds

3vangel1st, 2022-07-14:
Hi, I'm using IDA Pro 7.7 with SDK 7.7 (on Windows) and used your latest builds to install. When loading this firmware sample from Asus: https://github.com/3vangel1st/FWare/blob/main/X515JAAS.306 , I do see it gets loaded. When starting the efiXplorer plugin from the menu, I click through the pop-ups and it starts running and doing some analysis. However it never completes since after a while IDA crashes and generates a mini-dump.

Any idea what possibly is going wrong? I admit this is the only firmware sample I've used so far ;)

yeggor, 2022-07-14:
Hi. I just checked. Can not reproduce this issue. It's working fine for me:

[image]

If you are using IDA without a Hex-Rays decompiler, it is best to build the plugin from master with Hex-Rays analysis disabled. I'll PM you to clarify some details.

3vangel1st, 2022-07-20:
Understood - don't have the Hex-Rays decompiler.

skochinsky, 2022-12-16:
@yeggor could you please fix the code to not invoke decompiler APIs if the decompiler is not present (e.g. `init_hexrays_plugin()` returned false)? We're getting crash reports from customers because the plugin calls nullptr.

yeggor, 2022-12-16:
@skochinsky thank you. It fixed now

yeggor, 2022-12-16:
I've tested it on x86 and arm EFI modules. Works without problems now. Sorry for the inconvenience.

```
Python>ida_hexrays.init_hexrays_plugin()
False
Python>ida_loader.load_and_run_plugin("efiXplorer", 1)
...
[efiXplorer] the log is saved in a JSON file
True
```

closed 2022-12-16 by yeggor