REhints/efiXplorer #80: I encountered some err when compiling efiXloader using the shell of visual studio 2017 community on my windows 10.

risuxx, 2023-03-07:
The following content is the input and output displayed in the shell.
```
D:\efiXplorer\efiXloader\build>cmake .. -DIdaSdk_ROOT_DIR="D:\efiXplorer\SDK75\idasdk75"
-- Building for: Visual Studio 15 2017
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.16.27049.0
-- The CXX compiler identification is MSVC 19.16.27049.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found IdaSdk: D:/efiXplorer/SDK75/idasdk75
-- Configuring done (3.4s)
-- Generating done (0.1s)
-- Build files have been written to: D:/efiXplorer/efiXloader/build

D:\efiXplorer\efiXloader\build>cmake --build . --config Release
用于 .NET Framework 的 Microsoft (R) 生成引擎版本 15.9.21+g9802d43bc3
版权所有(C) Microsoft Corporation。保留所有权利。

  1>Checking Build System
  1>Building Custom Rule D:/efiXplorer/efiXloader/CMakeLists.txt
cl : 命令行 warning D9002: 忽略未知选项“/EH;” [D:\efiXplorer\efiXloader\build\efiXloader_lib_ea64.vcxproj]
cl : 命令行 warning D9002: 忽略未知选项“/EH-” [D:\efiXplorer\efiXloader\build\efiXloader_lib_ea64.vcxproj]
  efiLoader.cpp
cl : 命令行 warning D9002: 忽略未知选项“/EHf” [D:\efiXplorer\efiXloader\build\efiXloader_lib_ea64.vcxproj]
cl : 命令行 warning D9002: 忽略未知选项“/EHP” [D:\efiXplorer\efiXloader\build\efiXloader_lib_ea64.vcxproj]
cl : 命令行 warning D9002: 忽略未知选项“/EHI” [D:\efiXplorer\efiXloader\build\efiXloader_lib_ea64.vcxproj]
cl : 命令行 warning D9002: 忽略未知选项“/EHC” [D:\efiXplorer\efiXloader\build\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\pro.h(3191): warning C4305: “=”: 从“unsigned __int64”到“size_t”截断 [D:\efiXplorer\efi
Xloader\build\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\pro.h(3191): warning C4309: “=”: 截断常量值 [D:\efiXplorer\efiXloader\build\efiXloader_
lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\pro.h(3311): warning C4305: “=”: 从“unsigned __int64”到“size_t”截断 [D:\efiXplorer\efi
Xloader\build\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\pro.h(3311): warning C4309: “=”: 截断常量值 [D:\efiXplorer\efiXloader\build\efiXloader_
lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\idp.hpp(337): error C2338: sizeof(asm_t) == 408 [D:\efiXplorer\efiXloader\build\ef
iXloader_lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\idp.hpp(1864): error C2338: sizeof(processor_t) == 144 [D:\efiXplorer\efiXloader\b
uild\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\loader.hpp(609): error C2338: sizeof(plugin_t) == 64 [D:\efiXplorer\efiXloader\bui
ld\efiXloader_lib_ea64.vcxproj]
d:\efixplorer\efixloader\pe.h(111): warning C4018: “<”: 有符号/无符号不匹配 [D:\efiXplorer\efiXloader\build\efiXloader_lib_ea64.
vcxproj]
D:\efiXplorer\efiXloader\efiLoader.cpp(95): warning C4244: “参数”: 从“int64”转换到“size_t”，可能丢失数据 [D:\efiXplorer\efiXloader\b
uild\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\efiXloader\efiLoader.cpp(97): warning C4244: “参数”: 从“int64”转换到“size_t”，可能丢失数据 [D:\efiXplorer\efiXloader\b
uild\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\efiXloader\efiLoader.cpp(107): warning C4244: “参数”: 从“int64”转换到“size_t”，可能丢失数据 [D:\efiXplorer\efiXloader\
build\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\efiXloader\efiLoader.cpp(108): warning C4244: “参数”: 从“int64”转换到“size_t”，可能丢失数据 [D:\efiXplorer\efiXloader\
build\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\efiXloader\efiLoader.cpp(132): warning C4018: “<”: 有符号/无符号不匹配 [D:\efiXplorer\efiXloader\build\efiXloader_
lib_ea64.vcxproj]
  pe.cpp
D:\efiXplorer\SDK75\idasdk75\include\pro.h(3191): warning C4305: “=”: 从“unsigned __int64”到“size_t”截断 [D:\efiXplorer\efi
Xloader\build\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\pro.h(3191): warning C4309: “=”: 截断常量值 [D:\efiXplorer\efiXloader\build\efiXloader_
lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\pro.h(3311): warning C4305: “=”: 从“unsigned __int64”到“size_t”截断 [D:\efiXplorer\efi
Xloader\build\efiXloader_lib_ea64.vcxproj]
D:\efiXplorer\SDK75\idasdk75\include\pro.h(3311): warning C4309: “=”: 截断常量值 [D:\efiXplorer\efiXloader\build\efiXloader_
lib_ea64.vcxproj]
[... 89 lines cut ...]
  ffsreport.cpp
  ffsutils.cpp
  fitparser.cpp
  guiddatabase.cpp
  meparser.cpp
  nvram.cpp
  nvramparser.cpp
  peimage.cpp
  treeitem.cpp
  正在生成代码...
  正在编译...
  treemodel.cpp
  types.cpp
  ustring.cpp
D:\efiXplorer\efiXloader\3rd\uefitool\common\ustring.cpp(85): warning C4996: '_vsnprintf': This function or variable ma
y be unsafe. Consider using _vsnprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help
for details. [D:\efiXplorer\efiXloader\build\efiXloader_lib_ea64.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdio.h(1400): note: 参见“_vsnprintf”的声明
  utility.cpp
  正在生成代码...

D:\efiXplorer\efiXloader\build>
```
The err message seems from  some CASSERT sentence in the idasdk, but I don't know how to fix it.
There is not efiXloader.dll in D:\efiXplorer\efiXloader\build and D:\efiXplorer\efiXloader\build\Release. It failed to compile and generate the dll file.

yeggor, 2023-03-14:
I think you are trying to build a 32 bit target instead of 64-bit.
This is why `sizeof(TYPE)` does not match (because ida expects a pointer size of 64 bits instead of 32).

Apart from that I see that you are building a plugin with idasdk75. In this case, you will have to select a project state that is compatible with idasdk75. As the current sdk version supported by efiXplorer is idasdk82.

yeggor, 2023-03-14:
I checked and made sure that if I'll add `-m32` to compiler flags and will run `cmake` with `-A Win32`, I have the same issues. So you should check your setup and find out why you are building a 32-bit target instead of a 64-bit.

closed 2023-03-14 by yeggor