REhints/efiXplorer #84: [Suggestion] Top-level Cmake build file

NikolajSchlej, 2023-12-17:
It's somewhat tedious to run the very same build command twice, and it's trivial to fix: add this to CMakeLists.txt into the project root folder:
```
CMAKE_MINIMUM_REQUIRED(VERSION 3.7)

PROJECT(efiXplorer_everything)

ADD_SUBDIRECTORY(efiXplorer)
ADD_SUBDIRECTORY(efiXloader)
```

yeggor, 2023-12-20:
yeah, that definitely makes sense, thank you. I will add it now

yeggor, 2023-12-20:
it will build plugin and loader now:
```
mkdir build && cd build
cmake .. -DIdaSdk_ROOT_DIR=~/sdk/idasdk83 -DHexRaysSdk_ROOT_DIR=~/sdk/hexrays_sdk
cmake --build . --config Release
```

or with `build.py`:
```
python3 build.py build-everything --hexrays_sdk=~/sdk/hexrays_sdk ~/sdk/idasdk83
```

closed 2023-12-20 by yeggor

NikolajSchlej, 2023-12-20:
Might need an update to reflect the change here: https://github.com/binarly-io/efiXplorer/wiki/Build-instruction-and-installation

yeggor, 2023-12-20:
thank you, updated