REhints/efiXplorer #93: Crash 安天

gandalf4a, 2024-10-07:
When I tested arm UEFI, both PEI and DXE modes crashed when executing fixOffsets(). How to deal with this problem? The following is a dump file.

gandalf4a, 2024-10-07:
[AcpiTableDxe.zip](https://github.com/user-attachments/files/17281526/AcpiTableDxe.zip)

yeggor, 2024-10-07:
Hi. I see that you are using IDA 83 when latest version of efiXplorer was not compatible with it. I made it compatible and fixed some issues I didn't notice during the big change when updating the plugin and porting it to IDA 90.

Could you please use the latest version from master:

```
python3 build.py build-plugin ~/sdk/83/idasdk83 --hexrays_sdk ~/sdk/83/hexrays_sdk
```

I've tested your test case with IDA 83 and newer versions of IDA and can confirm that it's working:

[image]

As for the crash in `fixOffsets()` (which is now `fix_offsets()`), I can't see how this is possible. I guess it can only happen if the sdk and IDA version are incompatible. Please note that you'll have to build efixplorer with corresponding sdk.

gandalf4a, 2024-10-08:
Thanks for compatibility with ida83. I used the master version to compile through, and did not crash, but no matter the PEI or DXE module will appear unresponded pop-up window:
[image]
What is the cause of this situation? My cpu and memory usage are only about 40%.  I don't think it's due to poor hardware performance.
[image]

yeggor, 2024-10-08:
could you share binary on which you catched this? Maybe somewhere we didn't make an early return and it went unnoticed during testing (as arm support was tested much weaker than x86)

yeggor, 2024-10-08:
I've added potential fix (which might affect analysis for DXE modules). Сould you check please? And still share the binary or firmware if possible?

gandalf4a, 2024-10-08:
The firmware I tested was the same as the attachments above ‘AcpiTableDxe.efi’. 
Here's what I compiled the master（https://github.com/binarly-io/efiXplorer/commit/ef4d5e17560ae39417b5e2ce8b9a1a775d710d56) with the 83sdk, and I just added msg output to some functions to figure out what went wrong:
[efiXplorer64.zip](https://github.com/user-attachments/files/17288745/efiXplorer64.zip)
Below is a picture of me executing other efi files, as follows:
[ArmCpuDxe.zip](https://github.com/user-attachments/files/17288814/ArmCpuDxe.zip)
You can see that there is still no response when the fix_Offsets function is executed:
[image]

gandalf4a, 2024-10-08:
I added some msg output to fix_offsets() like this:
[image]
As you can see from the ida output, you should be stuck in a while loop and unable to exit, losing response.
[image]

gandalf4a, 2024-10-08:
I made a mistake, not losing the response at fix_offsets(), but at fix_offsets(). 

Here's what I tested by adding some msg：
[image]

You can see that it executes into the for loop and the following code is unresponsive:
```
    set_name(ep, "_ModuleEntryPoint", SN_FORCE);
#ifdef HEX_RAYS
    efi_hexrays::track_entry_params(get_func(ep), 0);
#endif /* HEX_RAYS */
```

[image]

gandalf4a, 2024-10-08:
I made sure to use the hexrays sdk.

However, this item is not displayed when cmake is configured.
[image]

Did it not work?

gandalf4a, 2024-10-09:
A close examination of the compiled output below shows that hexrays_sdk does indeed refer to it.

[image]

In addition, I commented out the above line of code, but the plugin execution still loses response when other hexrays_sdk functions are executed.

There are about a dozen plug-ins in my ida, but they are not implemented. Will this affect me?

gandalf4a, 2024-10-14:
I tested IDA 9.0 version and didn't encounter any unresponsive bugs, which should mean there's something wrong with my 8.3 version. I can close this issue. 

However, I encountered two errors while compiling the 9.0 version, which I fixed by following the instructions provided.
```
/home/gandalf/test/005-idapython/efiXplorer/efiXplorer/efi_utils.cc: In function ‘ea_list_t efi_utils::search_protocol(std::string)’:
/home/gandalf/test/005-idapython/efiXplorer/efiXplorer/efi_utils.cc:621:9: error: ‘bin_search’ was not declared in this scope; did you mean ‘bin_search3’?
  621 |         bin_search(start, BADADDR, bytes, nullptr, 16, BIN_SEARCH_FORWARD);
      |         ^~~~~~~~~~
      |         bin_search3
/home/gandalf/test/005-idapython/efiXplorer/efiXplorer/efi_utils.cc: In function ‘ea_list_t efi_utils::find_data(ea_t, ea_t, uchar*, size_t)’:
/home/gandalf/test/005-idapython/efiXplorer/efiXplorer/efi_utils.cc:850:15: error: ‘bin_search’ was not declared in this scope; did you mean ‘bin_search3’?
  850 |     auto ea = bin_search(start, end_ea, data, nullptr, len, BIN_SEARCH_FORWARD);
      |               ^~~~~~~~~~
      |               bin_search3
```

xorpse, 2024-10-14:
The compilation errors you have for IDA 9.0 are due to differences between the beta SDK and release SDK (Hex-Rays renamed the `bin_search` API to `bin_search3` in 9.0 RC1). You should update your IDA installation and SDK to the latest version.

yeggor, 2024-10-14:
I agree with @xorpse. For IDA 90 we only support the latest released version (and corresponding sdk), and `bin_search3` was used in beta versions prior to RC1. I'm closing the issue. Feel free to reopen it if you will have any other related issues that we can help you with.

closed 2024-10-14 by yeggor