REhints/efiXplorer #106: "error: The specified procedure could not be found." on IDA startup with v6.1 on IDA 8.3

tandasat, 2025-02-08:
I am unable to use efiXplorer v6.1 on my IDA 8.3 setup. It appears that efiXplorer64.dll now depends on API that is not available on 8.3 SDK, as I see this log message on startup of of IDA after updating efiXplorer from 6.0.

[image]

Since v6.1 is still supposed to support IDA SDK v8.3, I would appreciate if you can take a look and fix it. Thank you!

yeggor, 2025-02-08:
Hi. I've checked and I can see that efiXplorer is working with IDA 8.3:

```
...
      ____ _  __     __
 ___ / _(_) |/_/__  / /__  _______ ____
/ -_) _/ />  </ _ \/ / _ \/ __/ -_) __/
\__/_//_/_/|_/ .__/_/\___/_/  \__/_/
            /_/
(C) 2020-2024  Binarly - https://github.com/binarly-io/efiXplorer

Hex-rays version 8.3.0.230608 has been detected, gooMBA plugin ready to use
...
```

However, this requires rebuilding it with idasdk/hexrays_sdk for IDA 8.3.

If you use the builds that came with the release, they will only work on the latest version and are not backward compatible.

yeggor, 2025-02-08:
I've attached a build that should work for you (I can't test the 83 version on Windows at the moment, so let me know if you have any problems with it): [efixplorer-83.tar.gz](https://github.com/user-attachments/files/18721972/efixplorer-83.tar.gz)

```
4fe857c82df794fb30343abe12f7ae0ff11d3e0613b67a8def0bb3f786d0648e  efixplorer-83.tar.gz

├── guids
│   └── guids.json
├── loader
│   └── efiXloader64.dll
└── plugin
    ├── efiXplorer.dll
    └── efiXplorer64.dll
```

If you'd like to rebuild it yourself, please mind the following comment: https://github.com/binarly-io/efiXplorer/blob/96bdd13e32b636b9561efb6abbe6972e7e5161d9/efiXplorer/CMakeLists.txt#L58C1-L59C33

tandasat, 2025-02-09:
Confirmed that the attached file worked! I did not know I needed to rebuild it. Thank you a lot for the quick response. Closing it

closed 2025-02-09 by tandasat