REhints/efiXplorer #114: Question: supported editions of IDA?

tdceus, 2025-04-08:
Hi,

Maybe a stupid question but which editions of IDA are supported by the efiXplorer plugin?
Is IDA Free (PC: x86 32 & x86 64) or IDA Home supported?

Thank you.

Kind regards,
Tom

yeggor, 2025-04-08:
Hi Tom, thanks for the question (it's a good one, and the answer should ideally be in the readme/wiki in the efiXplorer repository).

`efiXplorer` is currently developed and tested with IDA Pro and the corresponding SDK for IDA Pro (minimum tested version is `v8.3`). However, there's no restriction on using it with other editions. Of course, `efiXplorer` for IDA Pro will offer better analysis quality and support for more architectures. This is mainly because the plugin contains features based on the Hex-Rays SDK, so the quality of the analysis and the quality of the pseudocode will be worse without native decompiler support.

However, I just tested the plugin with IDA Free (preliminarily compiled it with `idasdkhome91`, which is distributed for free from `my.hex-rays.com`), and I can confirm that all the features that are supposed to work are working:

[image]

Even annotations for global variables and function arguments have been applied to the pseudocode generated by a cloud-based decompiler (you can see that we were able to rediscover the following security issue: https://github.com/binarly-io/Vulnerability-REsearch/blob/main/HP/BRLY-2021-003.md).

tdceus, 2025-04-09:
Hi Yeggor, thanks for your feedback.

This answered my question if the plugin could be used with the free edition. :) I'm not a professional user.
I tried to analyse the firmware of my Lenovo's laptop for the hidden 'advanced' menu. But during the analysis IDA Free crashes. Maybe I should log a new incident here? I don't know if it problem is related to IDA or the plugin.

But for this question, you can close this issue.

yeggor, 2025-04-09:
Which module from `BIOS.fd` did you tried to analyse? As far as I understand, it's a Lenovo updater, it contains EFI installer and the firmware glued to it. If you will open `BIOS.fd` in IDA, it will open the EFI application for firmware flashing (but not the firmware or modules from firmware). To extract the modules for analysis, you will need to use [UEFITool](https://github.com/LongSoft/UEFITool) or analogues. [efiXloader](https://github.com/binarly-io/efiXplorer/tree/master/efiXloader) will allow to load `BIOS.fd` as a UEFI firmware, so it will load all the modules from it in a single `idb`, not just the flash application (but unfortunately it will not work with IDA Free).

Also, do I understand correctly that you didn't rebuild the plugin from master before using it, but installed it from the release directory? If so, try the version from build artefacts (https://github.com/binarly-io/efiXplorer/actions/runs/14341998102), but better rebuild yourself with the `idasdkhome91` sdk to make sure everything is compatible.

yeggor, 2025-04-09:
[BIOS.fd.modules.zip](https://github.com/user-attachments/files/19671380/BIOS.fd.modules.zip) -- here are modules from the attached firmware.

I was not able to reproduce the crash on mac, so I assume this is the fixed problem I wrote about, and you need to use the updated version of the plugin (not from release from build artefacts or build it yourself with `idasdkhome91`).

tdceus, 2025-04-10:
Hello Yeggor,

Thank you very much. I completely missed the fact that there was an installer wrapped around. I forgot, didn't know. Extracting the image from it gives a more clear result. :) Sorry, about that. Secondly I used indeed the latest release version. Now extracting a module in UEFITool like SetupUtility. Analyze that in IDE Free works with the latest build of efiExplorer. I just checked it.

Thanks for your valueable input. Now, I gonna try learning more about UEFI images.

yeggor, 2025-04-10:
Hi Tom,

I'm glad it worked out for you in the end. Good luck with your research!

closed 2025-04-10 by yeggor