REhints/efiXplorer #116: No show in Plugin Menu

chenzhaoting, 2025-05-09:
After Copy GUIDs  floder and efiXplore.dll   efiXplore64.dll to plugins,and efiXLoader.dll to loaders folder,restart IDA not show Exfxplore in Plugin menu,When  Open efi File No Exsit in Menu too.

yeggor, 2025-05-09:
is there any information related to efixplorer in the output window?

[image]

If the plugin has been successfully installed, you should see the same thing as in the screenshot. Otherwise, you should see errors (if, for example, efixplorer build is not compatible ida version that you are using).

Please provide more information if you would like me to help you sort this out:
- which ida version/edition are you using?
- which efixplorer version are you using?

chenzhaoting, 2025-05-09:
use older version,and open it with PE.dll,metaPC,it‘s work now.

closed 2025-05-09 by chenzhaoting